jiangcuo/pxvirt #117: 鲲鹏920服务器8.1.7新建虚拟机报错,8.2.4无法安装

2314933036, 2024-08-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: [e.g. debian11]
 - ARCH: [e.g. arm64,]
 - Version [ do `pveversion -v`]
 ```
proxmox-ve: 8.1.0 (running kernel: 6.1.73-generic)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.73-generic: 6.1.73-1
pve-kernel-6.1-rockchip: 6.1.43-1
pve-kernel-6.1.43-rockchip: 6.1.43-1
pve-kernel-5.15.145-generic: 5.15.145-1
pve-kernel-5.10.0-openeuler: 5.10.0-1
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.25-1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.6
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: 4.2023.08-4
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port2
smartmontools: 7.3-1+b1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.0-pve4

```

**Additional context**
Add any other context about the problem here.

2314933036, 2024-08-21:
[image]

2314933036, 2024-08-22:
虚拟机配置如图
[image]
[image]

yangzhenyu-123, 2024-08-28:
openuler2203也遇到这个问题，2003能正常安装

closed 2025-04-23 by jiangcuo