jiangcuo/pxvirt #119: error after start new VM `kvm: Failed to put registers after init: Invalid argument`

furyhawk, 2024-08-30:
**Describe the bug**
The rockpi 5 pro isn't support I guess, hope you can help.

error after start new VM:
```
kvm: Failed to put registers after init: Invalid argument
TASK ERROR: start failed: QEMU exited with code 1
```

**To Reproduce**
create VM:
```
agent: 1
boot: order=scsi0;scsi2;net0
cores: 2
memory: 2048
meta: creation-qemu=9.0.0,ctime=1725021860
name: default-vm
net0: virtio=BC:24:11:F1:29:51,bridge=vmbr0,firewall=1
numa: 0
ostype: l26
scsi0: local:902/vm-902-disk-0.qcow2,discard=on,iothread=1,size=16G,ssd=1
scsi2: local:iso/debian-12.6.0-arm64-netinst.iso,media=cdrom,size=538924K
scsihw: virtio-scsi-single
smbios1: uuid=ae7210c5-6dc3-4fa1-b687-5d28999ea836
sockets: 1
```
**ENV (please complete the following information):**
 - OS:
 ```
  ___  ____  _   ____    ____
 / _ \|  _ \(_) | ___|  |  _ \ _ __ ___
| | | | |_) | | |___ \  | |_) | '__/ _ \
| |_| |  __/| |  ___) | |  __/| | | (_) |
 \___/|_|   |_| |____/  |_|   |_|  \___/

Welcome to Armbian 24.5.2 Bookworm with Linux 6.1.75-vendor-rk35xx
```
 - ARCH: arm64
 - Version [ do `pveversion -v`]
proxmox-ve: 8.2.0 (running kernel: 6.1.75-vendor-rk35xx)
pve-manager: 8.2.4 (running version: 8.2.4/0098b77b88c86db8)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.7
libpve-cluster-perl: 8.0.7
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-rs-perl: 0.8.9
libpve-storage-perl: 8.2.1
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: not correctly installed
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 9.0.2-1
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0

I started CT successfully after removing NetworkManager used by Armbian.

furyhawk, 2024-08-30:
just saw~~~:
```
pve-edk2-firmware: not correctly installed
pve-firmware: not correctly installed
```

shvetsnikita, 2024-09-06:
Did you tried start vm with just one core?

furyhawk, 2024-09-09:
> Did you tried start vm with just one core?

try with 1 core, same error msg. Using Raspberry PI 5 is good so far.

memcoreDE, 2024-09-11:
Same issue on my Orange Pi 5 Plus 16GB with `Orange Pi 1.0.8 Bookworm with Linux 6.1.43-rockchip-rk3588`

```
kvm: Failed to put registers after init: Invalid argument
TASK ERROR: start failed: QEMU exited with code 1
```

memcoreDE, 2024-09-13:
Since I defined the "CPU Affinity" setting (Advanced), the VM starts without this error.

furyhawk, 2024-09-15:
it worked after setting `CPU Affinity`: 4,5,6,7

closed 2024-09-15 by furyhawk