jiangcuo/pxvirt #120: After passinc device, lxc won't boot

Anagastes, 2024-09-06:
**Describe the bug**
I'm followd [this](https://mattm.nz/blog/raspotify-proxmox-lxc/) to passthrough the audio device to lxc.

after it, the lxc didnt boot with latest pve and kernel

**To Reproduce**
Follow the wiki with pct set ^^ 

**Expected behavior**
lxc boot with sound devices :) 

**ENV (please complete the following information):**
 - OS: debian12
 - ARCH: arm64
 - Version:

proxmox-ve: 8.2.0 (running kernel: 6.6.31+rpt-rpi-v8)
pve-manager: 8.2.4 (running version: 8.2.4/0098b77b88c86db8)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.7
libpve-cluster-perl: 8.0.7
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-network-perl: 0.9.8
libpve-rs-perl: 0.8.9
libpve-storage-perl: 8.2.1
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: not correctly installed
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 9.0.2-1
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.4-pve1

**Additional context**
I will help to debug, but in the normal debug output only this is found...

```
buffer:571 - Script exited with status 2
Failed to run lxc.hook.pre-start for container "104"
```

tiehfood, 2024-09-06:
Same issue here

mgranberry, 2024-11-23:
```
lxc-start -n 128 -F -lDEBUG -o lxc-101.log
ot@orange1:/tmp# cat lxc-101.log 
lxc-start 128 20241123045123.780 INFO     confile - ../src/lxc/confile.c:set_config_idmaps:2273 - Read uid map: type u nsid 0 hostid 100000 range 65536
lxc-start 128 20241123045123.780 INFO     confile - ../src/lxc/confile.c:set_config_idmaps:2273 - Read uid map: type g nsid 0 hostid 100000 range 65536
lxc-start 128 20241123045123.785 INFO     lsm - ../src/lxc/lsm/lsm.c:lsm_init_static:38 - Initialized LSM security driver AppArmor
lxc-start 128 20241123045123.785 INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxc/hooks/lxc-pve-prestart-hook" for container "128", config section "lxc"
lxc-start 128 20241123045124.669 DEBUG    utils - ../src/lxc/utils.c:run_buffer:560 - Script exec /usr/share/lxc/hooks/lxc-pve-prestart-hook 128 lxc pre-start produced output: https://github.com/jiangcuo/Proxmox-Arm64/issues/58 at /usr/share/perl5/PVE/Tools.pm line 1756.

lxc-start 128 20241123045124.700 ERROR    utils - ../src/lxc/utils.c:run_buffer:571 - Script exited with status 2
lxc-start 128 20241123045124.700 ERROR    start - ../src/lxc/start.c:lxc_init:845 - Failed to run lxc.hook.pre-start for container "128"
lxc-start 128 20241123045124.700 ERROR    start - ../src/lxc/start.c:__lxc_start:2034 - Failed to initialize container "128"
lxc-start 128 20241123045124.701 INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxcfs/lxc.reboot.hook" for container "128", config section "lxc"
lxc-start 128 20241123045125.206 INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxc/hooks/lxc-pve-poststop-hook" for container "128", config section "lxc"
lxc-start 128 20241123045126.466 ERROR    lxc_start - ../src/lxc/tools/lxc_start.c:lxc_start_main:307 - The container failed to start
lxc-start 128 20241123045126.467 ERROR    lxc_start - ../src/lxc/tools/lxc_start.c:lxc_start_main:312 - Additional information can be obtained by setting the --logfile and --logpriority options
```

PVE is expecting to be able to mknod the new devices, but that functionality got stripped in https://github.com/jiangcuo/Proxmox-Arm64/issues/58 and hasn't been reintroduced.

jiangcuo, 2025-04-23:
Please check the associated commit, modify the local file, and then conduct the test

nickles95, 2025-05-05:
> Please check the associated commit, modify the local file, and then conduct the test

I've just done so and I successfully started an LXC container with usb devices passed through to it:

[image]

Anagastes, 2025-05-05:
For me, its doesn't work. ^^

I've changed the mentioned files and reboot the whole rpi4

[image]

jiangcuo, 2025-05-05:
> For me, its doesn't work. ^^
> 
> I've changed the mentioned files and reboot the whole rpi4
> 
> [image]

Please provide more logs

Anagastes, 2025-05-05:
Failed with device:

```bash
root@px:~ # pct start 101 --debug
run_buffer: 571 Script exited with status 2
lxc_init: 845 Failed to run lxc.hook.pre-start for container "101"
__lxc_start: 2034 Failed to initialize container "101"
d 0 hostid 100000 range 65536
INFO     lsm - ../src/lxc/lsm/lsm.c:lsm_init_static:38 - Initialized LSM security driver nop
INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxc/hooks/lxc-pve-prestart-hook" for container "101", config section "lxc"
DEBUG    utils - ../src/lxc/utils.c:run_buffer:560 - Script exec /usr/share/lxc/hooks/lxc-pve-prestart-hook 101 lxc pre-start produced output: https://github.com/jiangcuo/Proxmox-Arm64/issues/58 at /usr/share/perl5/PVE/Tools.pm line 1792.

ERROR    utils - ../src/lxc/utils.c:run_buffer:571 - Script exited with status 2
ERROR    start - ../src/lxc/start.c:lxc_init:845 - Failed to run lxc.hook.pre-start for container "101"
ERROR    start - ../src/lxc/start.c:__lxc_start:2034 - Failed to initialize container "101"
INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxc/hooks/lxc-pve-poststop-hook" for container "101", config section "lxc"
startup for container '101' failed
```

Successfull without:

```bash
root@px:~ # pct start 101 --debug
INFO     confile - ../src/lxc/confile.c:set_config_idmaps:2273 - Read uid map: type u nsid 0 hostid 100000 range 65536
INFO     confile - ../src/lxc/confile.c:set_config_idmaps:2273 - Read uid map: type g nsid 0 hostid 100000 range 65536
INFO     lsm - ../src/lxc/lsm/lsm.c:lsm_init_static:38 - Initialized LSM security driver nop
INFO     utils - ../src/lxc/utils.c:run_script_argv:587 - Executing script "/usr/share/lxc/hooks/lxc-pve-prestart-hook" for container "101", config section "lxc"
INFO     cgfsng - ../src/lxc/cgroups/cgfsng.c:unpriv_systemd_create_scope:1498 - Running privileged, not using a systemd unit
DEBUG    seccomp - ../src/lxc/seccomp.c:parse_config_v2:664 - Host native arch is [3221225655]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "reject_force_umount  # comment this to allow umount -f;  not recommended"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:532 - Set seccomp rule to reject force umounts
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:532 - Set seccomp rule to reject force umounts
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "[all]"
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "kexec_load errno 1"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[104:kexec_load] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[104:kexec_load] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "open_by_handle_at errno 1"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[265:open_by_handle_at] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[265:open_by_handle_at] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "init_module errno 1"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[105:init_module] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[105:init_module] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "finit_module errno 1"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[273:finit_module] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[273:finit_module] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "delete_module errno 1"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[106:delete_module] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[106:delete_module] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "ioctl errno 1 [1,0x9400,SCMP_CMP_MASKED_EQ,0xff00]"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:555 - arg_cmp[0]: SCMP_CMP(1, 7, 65280, 37888)
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[29:ioctl] action[327681:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:555 - arg_cmp[0]: SCMP_CMP(1, 7, 65280, 37888)
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[29:ioctl] action[327681:errno] arch[1073741864]
INFO     seccomp - ../src/lxc/seccomp.c:parse_config_v2:815 - Processing "keyctl errno 38"
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding native rule for syscall[219:keyctl] action[327718:errno] arch[0]
INFO     seccomp - ../src/lxc/seccomp.c:do_resolve_add_rule:572 - Adding compat rule for syscall[219:keyctl] action[327718:errno] arch[1073741864]
[... cut ...]
NOTICE   conf - ../src/lxc/conf.c:lxc_setup:4014 - The container "101" is set up
DEBUG    terminal - ../src/lxc/terminal.c:lxc_terminal_peer_default:709 - No such device - The process does not have a controlling terminal
NOTICE   start - ../src/lxc/start.c:start:2201 - Exec'ing "/sbin/init"
NOTICE   start - ../src/lxc/start.c:post_start:2212 - Started "/sbin/init" with pid "86322"
NOTICE   start - ../src/lxc/start.c:signal_handler:447 - Received 17 from pid 86323 instead of container init 86322
```

Anagastes, 2025-05-05:
> > For me, its doesn't work. ^^
> > I've changed the mentioned files and reboot the whole rpi4
> > [image]
> 
> Please provide more logs

Sorry, my bad... T_T simple syntax error.... i've fixed it and now it works -.-

closed 2025-05-08 by jiangcuo