jiangcuo/pxvirt #124: pve-no-subscription的源中arm64架构支持已被移除

zhufeng, 2024-09-30:
**Describe the bug**
pve-no-subscription的源中arm64架构支持已被移除，只有amd64的deb包了。

**To Reproduce**
```
[root@TG225-PVE ~]# apt update
Hit:1 http://mirrors.ustc.edu.cn/proxmox/debian/pve bookworm InRelease
Hit:2 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Hit:3 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
Hit:4 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
Hit:5 https://mirrors.ustc.edu.cn/debian-security stable-security InRelease
Hit:6 https://mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
6 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'pve-no-subscription/binary-arm64/Packages' as repository 'http://mirrors.ustc.edu.cn/proxmox/debian/pve bookworm InRelease' doesn't support architecture 'arm64'
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/arm64/release-notes/ch-information.html#non-free-split
```

**Expected behavior**
有无备用能用的arm64源。

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: [e.g. debian11]
 - ARCH: [e.g. arm64,]
 - Version [ do `pveversion -v`]
```
[root@TG225-PVE ~]# pveversion -v
proxmox-ve: 8.2.0 (running kernel: 6.1.62-generic)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.62-generic: 6.1.62
pve-kernel-5.15.138-generic: 5.15.138
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-network-perl: 0.9.8
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: 3.8-1
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 9.0.2-1
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.4-pve1
```

**Additional context**
Add any other context about the problem here.

Mikec78660, 2024-10-04:
Just started getting this as well.

jiangcuo, 2024-10-05:
看看wiki吧，写的很明白

closed 2025-04-23 by jiangcuo