jiangcuo/pxvirt #131: arm64 pve 8.1.7 + win10 arm64 按照wiki教程设置，bios能启动，press any key后卡在pve logo

chaolmrrt, 2024-11-09:
[image]

是不支持windows10 arm64 么？

bios: ovmf
boot: order=scsi2;scsi0;net0
cores: 4
cpu: max
efidisk0: data0:110/vm-110-disk-0.qcow2,efitype=4m,size=64M
memory: 4096
meta: creation-qemu=8.1.5,ctime=1731124890
name: rrt
net0: virtio=BC:24:11:A3:EE:18,bridge=vmbr0,firewall=1
numa: 0
ostype: l26
sata0: local:iso/virtio-win-0.1.217.iso,media=cdrom,size=519096K
scsi0: data0:110/vm-110-disk-1.qcow2,size=60G
scsi2: data0:iso/SW_DVD9_Win_Pro_10_22H2_64ARM_ChnSimp_Pro_Ent_EDU_N_MLF_X23-19974.ISO,media=cdrom,size=5195846K
smbios1: uuid=fd60bb69-d43f-462b-9bd4-4195d0e4e013
sockets: 1
vga: ramfb

chubirdh, 2025-01-06:
I install Windows 10 succesfully before, but failed to start VM after PVE was upgraded.  Now, I have the same problem when install Windows 10. But I can install Windows 11 22H2 with the same configuration for VM。

chubirdh, 2025-01-31:
According to  [this thread](https://github.com/virtio-win/kvm-guest-drivers-windows/issues/177), I changed the configuration of gic-version and succeeded installing Windows 10 22H2 ARM in 2 different CPU:
1. Huawei Hi1616: qm set <vmid> --args "--args "-machine 'virt-7.2,gic-version=2'"
[image]

2. Huawei Kunpeng-920:  qm set <vmid> --args  "-machine 'virt-9.2,gic-version=3,its=off'"
[image]

Event Viewer, Powershell all function normally.
Virtualization is not capable in VM.

jiangcuo, 2025-07-20:
This issue has been resolved

closed 2025-07-20 by jiangcuo