jiangcuo/pxvirt #132: Update failure

nickles95, 2024-11-24:
**Describe the bug**
Running apt update is complaining that the repo signature is invalid

**To Reproduce**
Run apt-update
Get this:
W: GPG error: https://mirrors.apqa.cn/proxmox/debian/pve bookworm Release: The following signatures were invalid: BADSIG B859507D6B1F46D3 jiangcuo (this is for pvearm key) <jiangcuo@bingsin.com>
E: The repository 'https://mirrors.apqa.cn/proxmox/debian/pve bookworm Release' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Try rerunning the steps to add the repo and download a fresh key from the wiki, issue persists.

**Expected behavior**
It works

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: Bookworm
 - ARCH: arm64
 - Version 
proxmox-ve: 8.2.0 (running kernel: 6.6.62-current-bcm2712)
pve-manager: 8.2.7 (running version: 8.2.7/368fab46e51eafa1)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.4
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.7
libpve-cluster-perl: 8.0.7
libpve-common-perl: 8.2.3
libpve-guest-common-perl: 5.1.4
libpve-http-server-perl: 5.1.1
libpve-rs-perl: 0.8.10
libpve-storage-perl: 8.2.5
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-4
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.3
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: not correctly installed
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 9.0.2-3
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.4
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0

**Additional context**
Add any other context about the problem here.

thanhtantran, 2024-11-24:
Get the same error

`The following signatures were invalid: BADSIG B859507D6B1F46D3 jiangcuo (this is for pvearm key) <jiangcuo@bingsin.com>`

vlalab, 2024-11-24:
Same problem triying to do a clean install.

sbilque, 2024-11-24:
Same problem triying to do a clean install by using any mirrors.

WebOscar, 2024-11-24:
Worked yesterday but since today I get the same error.

thanhtantran, 2024-11-25:
if you still need to install pve on bookworm, you can add the repo as trusted repo like below

`echo 'deb [arch=arm64 trusted=yes] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port'>/etc/apt/sources.list.d/pveport.list`

then update and install as normal

Use at your risk!!!!!

jiangcuo, 2024-11-25:
The reason lies in the fact that I manually input the key during signing previously, but I forgot to do so. Now it's auto-signed.

closed 2024-11-26 by nickles95