jiangcuo/pxvirt #133: Failed to fetch debs on apt upgrade

droidlev, 2024-11-29:
File has unexpected size: qemu-server_8.3.0_arm64.deb from mirros.apqa.cn

Anyone else having issues updating from the apqa mirror?

droidlev, 2024-11-29:
Totally thought it was me.  Much appreciate the reply, @Patient-9. Thanks!

strozykowski, 2024-11-29:
I have failures for 4 libraries, with similar error messages when doing a fresh install:
```
E: Failed to fetch https://mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/port//binary-arm64/libzfs4linux_2.2.6-pve1_arm64.deb  File has unexpected size (198876 != 198844). Mirror sync in progress? [IP: 172.67.186.225 443]
   Hashes of expected file:
    - SHA256:f6b73eac669cc6f364f18f4b91bab949a2223e8352325650aa10d7ea45141e95
    - SHA1:be12c773e0e411144ccbcd97666201174fa6731e [weak]
    - MD5Sum:4bb538716983d1294d9cbfc144aab1e2 [weak]
    - Filesize:198844 [weak]
E: Failed to fetch https://mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/port//binary-arm64/libzpool5linux_2.2.6-pve1_arm64.deb  File has unexpected size (1125604 != 1125580). Mirror sync in progress? [IP: 172.67.186.225 443]
   Hashes of expected file:
    - SHA256:d8dd5af50e4b9c4bdd53bbd56f20db6c8f021d7b9488603b54a1c95b6fd3e29a
    - SHA1:e10d7ebb5c41b70348bcbc0a0bedb9c360652755 [weak]
    - MD5Sum:545351780dee9bd74141cb4d54eb2263 [weak]
    - Filesize:1125580 [weak]
E: Failed to fetch https://mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/port//binary-arm64/qemu-server_8.3.0_arm64.deb  File has unexpected size (288432 != 286796). Mirror sync in progress? [IP: 172.67.186.225 443]
   Hashes of expected file:
    - SHA256:db69fec32b76f386b5fd03c234c167ad329cce1a16e5f1ca688cae9d6e94eb91
    - SHA1:67f163891107405c381f84e36180c9b0086737e2 [weak]
    - MD5Sum:bc77bf4f8abd683dd09f700726416b0c [weak]
    - Filesize:286796 [weak]
E: Failed to fetch https://mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/port//binary-arm64/zfsutils-linux_2.2.6-pve1_arm64.deb  File has unexpected size (529596 != 530168). Mirror sync in progress? [IP: 172.67.186.225 443]
   Hashes of expected file:
    - SHA256:862cd83cf5b259a7db1f2f91156881567021ba09cd14d5b189693ed1a4abbdd7
    - SHA1:d2c37d065f9233d1ba73c421ca2242e124b80bf6 [weak]
    - MD5Sum:71a2e9559c1bd838804369825ebfc5c9 [weak]
    - Filesize:530168 [weak]
```

droidlev, 2024-11-29:
No necessarily happy that you are experiencing the same/similar error(s), @strozykowski, but I am happy that someone else posted. I'm trying to do a fresh install and keep hitting this wall.  Any thoughts on an alternate mirror?

strozykowski, 2024-11-29:
Not sure about any mirrors, but this may be related to Issue #132

droidlev, 2024-11-29:
@strozykowski Just took a look there as well, at least what I am seeing it appear to be unrelated...  But who knows.  I did go ahead and try the German Mirror (same error).

They are rsync from mirrors.apqa.cn

Korea: https://mirrors.apqa.cn

Hong Kong: https://hk.mirrors.apqa.cn

China: https://mirrors.lierfang.com

Germany: https://de.mirrors.apqa.cn

droidlev, 2024-11-29:
China: https://mirrors.lierfang.com/ - appears to be working.
> @strozykowski Just took a look there as well, at least what I am seeing it appear to be unrelated... But who knows. I did go ahead and try the German Mirror (same error).
>
> They are rsync from [...]

jiangcuo, 2024-11-30:
I'm sorry for the trouble, but I'm updating the version through automation in the near future, and there will be some exceptions.The downtime will not be long,

droidlev, 2024-12-04:
Don't every apologize for offering up your free time!  Thank you for what you are doing!

closed 2025-04-23 by jiangcuo