jiangcuo/pxvirt #137: There is no rockchip install in the proxmox-ve_8.2-4_arm64.iso

Pix13, 2024-12-04:
**Describe the bug**
Using the OrangePI 5 Plus with the UEFI Bootloader ( rev 0.9.1), i can start grub but there is no rockchip intall entry, only the Install Proxmox and Install Proxmox with Kernel 6.1, none of them working

**To Reproduce**
Steps to reproduce the behavior:
1. Boot a orangepi5plus on the latest arm64 iso plugged
2. Setup the UEFI settings for the device tree
3. Start the ISO
4. No install option works

**Expected behavior**
Installation of Proxmox

**ENV (please complete the following information):**
 - OS: UEFI 0.9.1
 - ARCH: arm64
 - Version [proxmox-ve_8.2-4_arm64.iso](https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.2-4_arm64.iso)

**Additional context**
Iso of 8.1 worked well with the rockchip entry, but is not available anymore, if possible i would use it

Pix13, 2024-12-04:
Oh ok, so would it be possible to provide the old 8.1 image that supports it ?

Pix13, 2024-12-04:
Otherwise i guess the new way would be to install a rockchip debian/armbian, add your repos and then install this way

Pix13, 2024-12-04:
Installing armbian-uefi in AHCI mode and then installing pve kernel and switching to dtb in the uefi settings allow to install afterwards PVE and PBS :)

wuganzhi, 2025-02-09:
> armbian-uefi

I encountered the same problem. Can you provide a mirror address for armbian-uefi

wuganzhi, 2025-02-09:
> Installing armbian-uefi in AHCI mode and then installing pve kernel and switching to dtb in the uefi settings allow to install afterwards PVE and PBS :)在 AHCI 模式下安装 armbian-uefi，然后安装 pve 内核并在 uefi 设置中切换到 dtb，允许之后安装 PVE 和 PBS :)
I encountered the same problem. Can you provide a mirror address for armbian-uefi

jpdacostaza, 2025-03-09:
Ok guys so what was the action taken to resolve this and what kernel is your PVE currently using?
are there any benefits to debian/armbian repo install or is the openeuler kenel fine.

dandycyclone, 2025-03-12:
@jpdacostaza, you may use any kernel, that is supported by your current SBC firmware. In case of _rockchip_ and UEFI it essentially means using one of x2 options: https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#in-device-tree-mode - I personally still suggest using _rockchip_ kernel instead of generic.

What is more, even though "_rockchip will no longer be maintained_", vendor specific kernels could be still found in @jiangcuo kernel's repo: https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md#kernel-repo

Example from a running system:
```
root@pve:~# pveversion 
pve-manager/8.3.3/bbba1c53a1a65b24 (running kernel: 6.1.43-rockchip)
```

wuganzhi, 2025-03-13:
> look when installing the kernel everything seems to work but console gets spammed with :查看安装内核时，一切似乎都正常，但控制台被冒号（:）信息刷屏 hdmirx0 = "/hdmirx-controller@fdee0000"; <--- this is the main issuehdmirx0 = "/hdmirx-controller@fdee0000"; <--- 这是主要问题 This ultimetly crashed the device., any advise how / which, what and where would be apreciated此最终导致设备崩溃，任何关于如何/哪个、什么和在哪里建议都将受到欢迎。 [@dandycyclone](https://github.com/dandycyclone) which debian did you use, the orange pi debian or plain debain..?您使用了哪个 Debian，是橙子派 Debian 还是普通 Debian？ I will test today with : Orangepi5plus_1.2.0_debian_bookworm_server_linux6.1.43.7z from vendor.今天我将使用供应商的 Orangepi5plus_1.2.0_debian_bookworm_server_linux6.1.43.7z 进行测试。 With this have you reveerted back to U-Boot or are you running the EDK2 EFi boot..使用此方法，您是否已回滚到 U-Boot 或您正在运行 EDK2 EFi 启动程序.. Did you hold your firmware, any info would be apprreciated, will post the results and outcome.您保留了固件吗？任何信息都将受到欢迎，我将发布结果和结果。

I used Armbian to install the operating system and then installed the PVE by adding a repository, and there is no problem with it at the moment.

jpdacostaza, 2025-03-13:
@wuganzhi can you perhaps advise which armbian image you used.. ?

wuganzhi, 2025-03-13:
> [@wuganzhi](https://github.com/wuganzhi) can you perhaps advise which armbian image you used.. ? the https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm If you are using SBC ,you can install a Armbian Bullseye.

I'm using Orange Pi 5, so I'm installing a https://dl.armbian.com/orangepi5/Bookworm_vendor_minimal image, and you can find a mini image for your hardware on Armbian's official website. Finally, refer to the tutorial here https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm to install the PVE.

jpdacostaza, 2025-03-13:
@wuganzhi much appreciated, will test now , thanks.
-- Did you take any other steos besides the tutorial https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm to install the PVE.
-- post install .. ?

wuganzhi, 2025-03-13:
> [@wuganzhi](https://github.com/wuganzhi) much appreciated, will test now , thanks. -- Did you take any other steos besides the tutorial https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm to install the PVE. -- post install .. ?

https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm-based installation, partial optimisation, and firewall configuration reference: https://foxi.buduanwang.vip/virtualization/pve/1902.html/

jpdacostaza, 2025-03-14:
Issue Resolved: Using :  EDK2 UEFI firmware for Rockchip RK3588 platforms https://github.com/edk2-porting/edk2-rk3588

NOTE: most of the steps are post Flashing the EDK2 Fiirmware to the SPI of the Orange Pi 5 Plus.

1. Changed ACPI / Device Tree and setting Config Table Mode to AHCI
This contradicts what it says in the guide below but works.
https://github.com/jiangcuo/Proxmox-Port/wiki/Rockchip-RK3588-UEFI-Setup

2. BelenaEtcher : proxmox-ve_8.2-4_arm64.iso
proxmox-ve_8.2-4_arm64.iso](https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.2-4_arm64.iso
At this point im thinking anyone will do except the test ISO.

3. Booted with USB at the Proxmox Grub loader 
Advanced -> 6.1 Kernel Generic.

4. Used the Non-Gui Installer and installed to NVME no issues at all basic straight forward install .
NOTE: Set your hostname in the installer when the IP settings come up.
IF you dont you will have issues changing this later and would be better to reinstall at this point as its quick.

5. Followed the steps for Custom Linux kernels for Proxmox VE arm64
Check you are in the v6.12 brach. https://github.com/jiangcuo/pve-port-kernel/tree/v6.12
This took about 3 Hours on the OP5Plus itself and eventually generated the debs 

[image]

6. Moved the debs to a new folder and ran : dpkg -i *
There was some dependency mismatched and propt a apt "install --fix-broken install"
do this and then re-install the debs dpkg -i * check if they installed correctly.

7. Rebooted still in ACPI / AHCI mode and aurprisingly it did not boot.
-- Switched back to the older Generic kernel in Grub Advanced 

8. Check if there are any dtb / dtbs in /boot and remove them
as it seems they have been built into the kernel when you do step 5 it checks the hardware when building.

9. Rebooted into Bios ESC or F2 and changed the 
ACPI / Device Tree and setting Config Table Mode to Device tree if correct it was mainline with GOP and load DTB.

10. Rebooted and It works, Fast with no issues at all 
-- Perfomed an apt update / upgrade as at thie point the PVE-Manager was still on 8.0.1 or something
It upgraded to 8.3.3 and also updated the kernel still seeing no bad effects 

11. Tested the modules for Rockchip they are included in the kernel build and seem to all be working

For now that was the workaround. 
Thanks for all the help and feedback.

<html>
<body>

8 x Rockchip RK3588 CPU @ 0.8GHz Rockchip RK3588 CPU @ 0.8GHz (1 Socket)
--
Kernel Version Linux 6.12.11-3-pve (2025-01-31)
Boot Mode EFI
Manager Version pve-manager/8.3.3/bbba1c53a1a65b24

</body>
</html>