jiangcuo/pxvirt #138: Boot failure on new Loongson 3C6000 platform

KFERMercer, 2024-12-06:
安装镜像在 3c6000 上无法启动内核。

我使用的安装镜像：https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.1-4-2-loong64.iso

附上内核启动日志报错部分，供参考：

<details>

```log
[  147.629983] CPU 17 Unable to handle kernel paging request at virtual address ffff800002078000, era == 9000000003bf3bec, ra == 9000000002a032dc
[  147.642707] Oops[#1]:
[  147.644960] CPU: 17 PID: 1006 Comm: modprobe Not tainted 6.6.0-openeuler #1
[  147.651879] Hardware name: LOONGSON 20240815-AT1C6T0-PPN/Loongson-AC612A0, BIOS Loongson-UDK2018-V4.0.05692-stable202402 07/18/24 16:38:46
[  147.664239] pc 9000000003bf3bec ra 9000000002a032dc tp 9000000113614000 sp 9000000113617bc0
[  147.672538] a0 ffff800002078000 a1 0000000000000000 a2 ffff80000207c000 a3 9000000104d7c088
[  147.680837] a4 fffffffffe000000 a5 ffff800000000000 a6 0000800000000000 a7 900000027ff90110
[  147.689137] t0 0000000000000000 t1 741c20e915d67c8f t2 ffffffffffffffdf t3 900000010009c010
[  147.697435] t4 ffff80000207c000 t5 9000000003d264c0 t6 900000000485c000 t7 90000000047c4000
[  147.705734] t8 ffff80000207bfff u0 0000000000000001 s9 0000000000003fff s0 ffffffffffffc000
[  147.714032] s1 0000000000000002 s2 0000000000000000 s3 ffff8000120d5000 s4 0000000000000007
[  147.722331] s5 ffff800002078000 s6 900000000470bc40 s7 9000000113617d48 s8 ffff8000120d5218
[  147.730630]    ra: 9000000002a032dc load_module+0x6e4/0x1aa0
[  147.736257]   ERA: 9000000003bf3bec __memset_fast+0x18/0x138
[  147.741884]  CRMD: 000000b0 (PLV0 -IE -DA +PG DACF=CC DACM=CC -WE)
[  147.748032]  PRMD: 00000004 (PPLV0 +PIE -PWE)
[  147.752360]  EUEN: 00000000 (-FPE -SXE -ASXE -BTE)
[  147.757121]  ECFG: 00071c1d (LIE=0,2-4,10-12 VS=7)
[  147.761881] ESTAT: 00020000 [PIS] (IS= ECode=2 EsubCode=0)
[  147.767331]  BADV: ffff800002078000
[  147.770791]  PRID: 0014d010 (Loongson-64bit, Loongson-3C6000)
[  147.776498] Modules linked in: uas usb_storage spi_loongson_pci spi_loongson_core
[  147.783938] Process modprobe (pid: 1006, threadinfo=00000000c47ce434, task=000000007abe8653)
[  147.792324] Stack : 9000000113617c38 0000000000000001 900000000484d2a0 0000000000000000
[  147.800283]         0000000000000016 000000000000001d 0000000000000001 9000000004050370
[  147.808241]         0000000000000123 0000000000000000 000000000000000e 9000000113617d48
[  147.816198]         ffff8000120d4000 ffff8000120d6998 0000000000000000 0000000000000000
[  147.824155]         0000000000000001 000055555824af68 0000000000000000 0000000000000001
[  147.832113]         0000000000000000 0000000000000000 0000000000000000 741c20e915d67c8f
[  147.840070]         0000000000000000 0000000000000002 9000000113617cd0 90000000042e6000
[  147.848027]         0000000000003198 000000007fffffff 90000001120f7c00 9000000002c00d58
[  147.855985]         0000000000000002 ffff8000120d4000 0000000000003198 741c20e915d67c8f
[  147.863942]         90000000042e6000 9000000103722e88 0000000000000250 900000000484d560
[  147.871898]         ...
[  147.874322] Call Trace:
[  147.874324] [<9000000003bf3bec>] __memset_fast+0x18/0x138
[  147.882111] [<9000000002a032dc>] load_module+0x6e4/0x1aa0
[  147.887474] [<9000000002a04730>] init_module_from_file+0x98/0xe0
[  147.893441] [<9000000002a04ba4>] sys_finit_module+0x204/0x2f0
[  147.899151] [<9000000003bf3108>] do_syscall+0x8c/0xb4
[  147.904170] [<90000000029410dc>] handle_syscall+0xbc/0x158
[  147.909620] 
[  147.911092] Code: 024024cc  44008d80  00109886 <29c00085> 02c02087  00820007  02ff00c8  6c002ce8  29c000e5 
[  147.920785] 
[  147.922260] ---[ end trace 0000000000000000 ]---
[  169.042803] IPMI message handler: version 39.2
[  172.677793] CPU 4 Unable to handle kernel paging request at virtual address ffff8000020c0000, era == 9000000003bf3bec, ra == 9000000002a032dc
[  172.690424] Oops[#2]:
[  172.692675] CPU: 4 PID: 1086 Comm: systemd-udevd Tainted: G      D            6.6.0-openeuler #1
[  172.701408] Hardware name: LOONGSON 20240815-AT1C6T0-PPN/Loongson-AC612A0, BIOS Loongson-UDK2018-V4.0.05692-stable202402 07/18/24 16:38:46
[  172.713767] pc 9000000003bf3bec ra 9000000002a032dc tp 9000000112c98000 sp 9000000112c9bbc0
[  172.722066] a0 ffff8000020c0000 a1 0000000000000000 a2 ffff8000020c4000 a3 9000000104b00050
[  172.730365] a4 fffffffffe000000 a5 ffff800000000000 a6 0000800000000000 a7 900000027ff90110
[  172.738664] t0 0000000000000000 t1 741c20e915d67c8f t2 ffffffffffffffdf t3 900000010009c010
[  172.746963] t4 ffff8000020c4000 t5 9000000003d264c0 t6 900000000485c000 t7 90000000047c4000
[  172.755261] t8 ffff8000020c3fff u0 0000000000000001 s9 0000000000003fff s0 ffffffffffffc000
[  172.763560] s1 0000000000000001 s2 0000000000000000 s3 ffff80001218d9c0 s4 0000000000000007
[  172.771859] s5 ffff8000020c0000 s6 900000000470bc40 s7 9000000112c9bd48 s8 ffff80001218db90
[  172.780157]    ra: 9000000002a032dc load_module+0x6e4/0x1aa0
[  172.785784]   ERA: 9000000003bf3bec __memset_fast+0x18/0x138
[  172.791410]  CRMD: 000000b0 (PLV0 -IE -DA +PG DACF=CC DACM=CC -WE)
[... 112 lines cut ...]
[  315.263248] [<900000000339803c>] tty_init_dev.part.0+0xb4/0x20c
[  315.269132] [<9000000003398754>] tty_open+0x438/0x474
[  315.274150] [<9000000002bc0080>] chrdev_open+0xc0/0x210
[  315.279342] [<9000000002bb4714>] do_dentry_open+0x1b0/0x534
[  315.284879] [<9000000002bce128>] path_openat+0xbd8/0xfe4
[  315.290158] [<9000000002bceef0>] do_filp_open+0xa8/0x154
[  315.295434] [<9000000002bb50cc>] do_sys_openat2+0x2d4/0x310
[  315.300970] [<9000000002bb7054>] sys_openat+0x4c/0xa0
[  315.305988] [<9000000003bf3108>] do_syscall+0x8c/0xb4
[  315.311009] [<90000000029410dc>] handle_syscall+0xbc/0x158
[  315.316459] 
[  315.317930] Code: 02df2339  1a01aaa5  038a2084 <29c1430c> 00150326  00109304  02fd00a5  573513d8  14000044 
[  315.327622] 
[  315.329100] ---[ end trace 0000000000000000 ]---
```

</details>

另附上完整内核日志，以备需要：[host_20241206_194341.log](https://github.com/user-attachments/files/18037799/host_20241206_194341.log)

猜测应该是欧拉的内核问题（直接装[openeuler 2403 lts](https://repo.openeuler.org/openEuler-24.03-LTS/ISO/loongarch64/openEuler-24.03-LTS-loongarch64-dvd.iso)也是启动不起来）

小小建议可以考虑换 aosc 或者 debian sid

我有 3c6000 的 bmc 可以在线开放给您供您测试，需要的话可以进一步联系。

jiangcuo, 2024-12-13:
可以测试下新的8.3版本
https://mirrors.lierfang.com/proxmox/isos/proxmox-ve-8.3-1_loong64-pretest.iso

ltkun, 2024-12-15:
我的平台是华硕主板的3a6000 试过最新的两个iso 都不能安装 进入安装GUI以后只有 previous reboot两个按钮 只能寻求其他方式了

jiangcuo, 2024-12-15:
> 我的平台是华硕主板的3a6000 试过最新的两个iso 都不能安装 进入安装GUI以后只有 previous reboot两个按钮 只能寻求其他方式了

华硕的是自带的网卡把，可能没有网卡或者没有检测到硬盘。用debug模式，进入之后，ip a看一下，没有网卡就不会有，按理说12代是支持网卡的

ltkun, 2024-12-15:
应该是网卡问题了 记得切换到终端下没认出来 目前装了个aosc
`aosc-4937135 [ ~ ] $ lspci
00:00.0 Host bridge: Loongson Technology LLC Hyper Transport Bridge Controller
00:00.1 Host bridge: Loongson Technology LLC Hyper Transport Bridge Controller (rev 01)
00:00.2 Host bridge: Loongson Technology LLC Device 7a20 (rev 01)
00:00.3 Host bridge: Loongson Technology LLC Device 7a30
00:03.0 Ethernet controller: Loongson Technology LLC Device 7a13
00:04.0 USB controller: Loongson Technology LLC OHCI USB Controller (rev 02)
00:04.1 USB controller: Loongson Technology LLC EHCI USB Controller (rev 02)
00:05.0 USB controller: Loongson Technology LLC OHCI USB Controller (rev 02)
00:05.1 USB controller: Loongson Technology LLC EHCI USB Controller (rev 02)
00:06.0 Multimedia video controller: Loongson Technology LLC LG100 GPU (rev 01)
00:06.1 VGA compatible controller: Loongson Technology LLC Display Controller (rev 02)
00:06.2 Audio device: Loongson Technology LLC Device 7a37
00:07.0 Audio device: Loongson Technology LLC HDA (High Definition Audio) Controller
00:08.0 SATA controller: Loongson Technology LLC SATA 3 AHCI Controller
00:09.0 PCI bridge: Loongson Technology LLC PCIe x4 Root Port
00:0a.0 PCI bridge: Loongson Technology LLC PCIe x1 Root Port
00:0b.0 PCI bridge: Loongson Technology LLC PCIe x1 Root Port
00:0c.0 PCI bridge: Loongson Technology LLC PCIe x1 Root Port
00:0d.0 PCI bridge: Loongson Technology LLC PCIe x4 Root Port
00:0f.0 PCI bridge: Loongson Technology LLC PCIe x16 Root Port
00:10.0 PCI bridge: Loongson Technology LLC PCIe x8 Root Port
00:13.0 PCI bridge: Loongson Technology LLC PCIe x8 Root Port
00:16.0 System peripheral: Loongson Technology LLC SPI Controller
00:19.0 USB controller: Loongson Technology LLC xHCI USB Controller
01:00.0 Ethernet controller: Motorcomm Microelectronics. YT6801 Gigabit Ethernet Controller (rev 03)`
华硕主板两块网卡 我看内核好像是欧拉的 估计没带

jiangcuo, 2024-12-15:
YT6801没有这个驱动，我尝试打个补丁，在8.3正式iso出来之前解决这个问题，还请你到时候测测

ltkun, 2024-12-15:
OK PVE一直是我装机首选 不管有用没用先装上哈 感谢为PVE普及做出的贡献！

ltkun, 2024-12-15:
> YT6801没有这个驱动，我尝试打个补丁，在8.3正式iso出来之前解决这个问题，还请你到时候测测

Ethernet controller: Loongson Technology LLC Device 7a13
双网卡还有一块是这个

ltkun, 2024-12-16:
> 第二版测试应该可以了，加入了龙芯dwmac和yt6801网卡驱动。 https://mirrors.lierfang.com/proxmox/isos/proxmox-ve-8.3-1_loong64-pretest2.iso

感谢 今天回去试试 看来龙芯的PVE用户不多哈 或者说华硕这块主板的玩家不多

ltkun, 2024-12-19:
[image]
还是点GUI安装以后出错……

jiangcuo, 2024-12-24:
这个问题已经修复了，重新下载iso即可

ltkun, 2024-12-24:
> 这个问题已经修复了，重新下载iso即可

顺利安装上华硕3a6000 就是启动的时候隐约见了之前的错误 不过能启动哈 谢谢大佬

closed 2025-01-20 by jiangcuo