jiangcuo/pxvirt #139: pve-kernel-6.6.0-2-2-openeuler fails to boot on TaiShan 200 (Model 2280)

nE0sIghT, 2024-12-09:
**Describe the bug**
System doesn't boot just black screen with blinking cursor

**To Reproduce**
1. apt-get dist-upgrade

**Expected behavior**
System boots fine

**ENV (please complete the following information):**
 - OS: pve-manager/8.3.1/fb48e850ef9dde27
 - ARCH: arm64
 - Version
```
proxmox-ve: 8.3.0 (running kernel: 6.6.0-openeuler)
pve-manager: 8.3.1 (running version: 8.3.1/fb48e850ef9dde27)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.6-openeuler: 6.6.0-2
pve-kernel-6.6.0-2-2-openeuler: 6.6.0-2
pve-kernel-6.6.0-openeuler: 6.6-20241113
pve-kernel-6.1.102-generic: 6.1.102
pve-kernel-6.1-generic: 6.1.102
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx11
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-5
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.1
libproxmox-rs-perl: 0.3.4
libpve-access-control: 8.2.0
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.10
libpve-cluster-perl: 8.0.10
libpve-common-perl: 8.2.9+port
libpve-guest-common-perl: 5.1.6
libpve-http-server-perl: 5.1.2
libpve-network-perl: 0.10.0+port
libpve-rs-perl: 0.9.0
libpve-storage-perl: 8.2.9
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-4
proxmox-backup-client: 3.3.0-2
proxmox-backup-file-restore: 3.3.0-2
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.6
proxmox-widget-toolkit: 4.3.2
pve-cluster: 8.0.10
pve-container: 5.2.2+port1
pve-docs: 8.3.1
pve-edk2-firmware: not correctly installed
pve-firewall: 5.1.0
pve-firmware: 3.8-1
pve-ha-manager: 4.0.6
pve-i18n: 3.2.4
pve-qemu-kvm: 9.0.2-3
pve-xtermjs: 5.3.0-3
qemu-server: 8.3.0+port2
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.9.0
vncterm: 1.8.0
zfsutils-linux: 2.2.6-pve1
```

**Additional context**
pve-kernel-6.6-openeuler works fine

jiangcuo, 2024-12-12:
New kernel 6.6.0-4 uploaded.

nE0sIghT, 2024-12-15:
It works, thanks!

closed 2024-12-15 by nE0sIghT