jiangcuo/pxvirt #141: ZFS not loading

LuisPuertollano, 2024-12-19:
Hello Jiangcuo, hello everyone,

I`ve followed all linked ressources as resolution but I'm still stucked ...is there any way to have a unified explanation?

https://github.com/jiangcuo/Proxmox-Port/wiki/ZFS-Installation
https://github.com/jiangcuo/Proxmox-Port/issues/5
https://github.com/jiangcuo/Proxmox-Arm64/issues/45
https://github.com/jiangcuo/Proxmox-Port/issues/2

I'm trying to install the ZFS functionaly in the Proxmox installation running on a CM3588 (Rockchip), but it seemno not load properly.

actual status is following, I'm triying to create a ZFS Pool for 4x4TB SSDs, all are recognised (I think properly), but when I try to create the pool, ZFS GUI doesn't load:
command '/sbin/zpool list -HpPLo name,size,alloc,free,frag,dedup,health' failed: exit code 1 (500)
[image]

[image]

[image]

Hopefully some can guide me a little bit

dp2010, 2024-12-30:
Hello,

looks like the kernel headers are missing (had the same problem with rk3588 sbc).

you can try:

1. remove the non working packages:
apt remove zfs-dkms zfsutils-linux

----------

2. install kernel headers:

apt update

apt upgrade

apt install linux-headers-vendor-rk35xx

--------

3. install zfs packages again:

apt install zfs-dkms zfsutils-linux

----------

that way it worked for me.

idlethread, 2025-01-01:
In your `/etc/apt/sources.list.d/pveport.list` file, add the following lines:

```
# Needed for ZFS
deb http://ftp.fi.debian.org/debian/ bookworm contrib non-free-firmware non-free
deb http://ftp.fi.debian.org/debian/ bookworm-updates contrib non-free-firmware non-free
deb http://security.debian.org/debian-security/ bookworm-security contrib non-free-firmware no
n-free
deb http://deb.debian.org/debian bookworm-backports main non-free-firmware
```

Then run,

```
$ sudo apt update
$ sudo apt install linux-headers-arm64 zfs-dkms zfsutils-linux
```

Press OK and wait for some time for the zfs module to be compiled and then run,

`$ sudo modprobe zfs`

LuisPuertollano, 2025-01-02:
Hello people, I followed your commands (erased old headers, etc) and the instructions under:
https://wiki.friendlyelec.com/wiki/index.php/CM3588#Install_Kernel_Headers

ZFS seems to be running.

thank you all, I think this topic can be closed :-)

closed 2025-01-02 by LuisPuertollano