jiangcuo/pxvirt #142: TASK ERROR: EFI base image '/usr/share/pve-edk2-firmware//AAVMF_CODE.ms.fd' not found

danmrossi, 2024-12-23:
Hi All

Installed latest updates for Raspberry Pi 5 running bookworm using apt-get update / apt-get dist-upgrade and rebooted the PVE on the RPI 5 and one of my VM's gets: TASK ERROR: EFI base image '/usr/share/pve-edk2-firmware//AAVMF_CODE.ms.fd' not found.

Start-Date: 2024-12-23  19:21:25
Commandline: apt-get dist-upgrade
Requested-By: pve01 (1000)
Install: pve-edk2-firmware-loongarch64:arm64 (4.2024.11-1, automatic), pve-edk2-firmware-riscv:arm64 (4.2024.11-1, automatic)
Upgrade: pve-edk2-firmware-ovmf:arm64 (4.2023.08-4, 4.2024.11-1), pve-edk2-firmware-legacy:arm64 (4.2023.08-4, 4.2024.11-1), pve-edk2-firmware-aarch64:arm64 (4.2023.08-4, 4.2024.11-1), pve-edk2-firmware:arm64 (4.2023.08-4, 4.2024.11-1)
End-Date: 2024-12-23  19:21:28

This one started ok:
[image]

This one didnt and only worked after i did the symlink:
[image]

I had set them both up at the same time and are identical in all settings anywhere and everywhere, the only difference is USB passthru to the VM in the first pic i said started up fine.

To fix i created a symlink from AAVMF_CODE.fd to AAVMF_CODE.ms.fd and the VM now starts without issue

jiangcuo, 2024-12-24:
Solved it with upgrade pve-edk2-firmware-aarch64 to 4.2024.11-2.

Rappelkiste98, 2024-12-26:
@jiangcuo With the newest upgrade for "pve-edk2-firmware-aarch64" VM's on Raspberry Pi 4B doesn't boot.

Error Messeage while booting: "synchronous exception at 0x000000009BF8A438"

With downgrade to Version pve-edk2-firmware-aarch64 = 4.2023.08-4 everything works fine.

Zorryno, 2024-12-27:
got the same problem on both Paspberry Pi 4B and 5

jiangcuo, 2024-12-28:
The new version 4.2024.11-3 has been released, please test it

Zorryno, 2024-12-28:
seems to be fixed
thank you

mathisgauthey, 2025-04-04:
Thanks lads, `sudo apt update && sudo apt install pve-edk2-firmware-aarch64` effectively fixed it for me.

closed 2025-04-23 by jiangcuo