jiangcuo/pxvirt #143: Unable to install because of unmet dependencies

totheogreat, 2024-12-25:
Unable to install because of unmet dependencies on raspberry 4 (8 gig), using 64 bit OS (see below). Follow the "Install Proxmox VE on Debian bookworm" wiki.

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

sudo apt install proxmox-ve postfix open-iscsi pve-edk2-firmware-aarch64
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 proxmox-ve : Depends: qemu-server but it is not installable
 pve-ha-manager : Depends: qemu-server (>= 8.0.2) but it is not installable
 pve-manager : Depends: qemu-server (>= 8.2.7) but it is not installable

RedX8, 2024-12-25:
I have just noticed the same issue on my pi5.

totheogreat, 2024-12-25:
Haha, so I am not alone...

Makes two noobs of us, but the url still works: [https://mirrors.apqa.cn/proxmox/debian/pve] "bookworm" and "port" are references to the image apt needs to look for. I still see them

CrimsonKnight13, 2024-12-25:
It would appear that dependencies on jiangcuo's side will need to be updated. It appears to be a shift in requirements from Raspberry Pi OS.

splifingate, 2024-12-25:
Me three (unfortunately).

Tried the DE mirror to no avail.

Luckily, I had other things to do, today ;)

HafizAhmadJamil, 2024-12-25:
add 1 more

same here 
`The following information may help to resolve the situation:

The following packages have unmet dependencies:
 proxmox-ve : Depends: qemu-server but it is not installable
 pve-ha-manager : Depends: qemu-server (>= 8.0.2) but it is not installable
 pve-manager : Depends: qemu-server (>= 8.2.7) but it is not installable
E: Unable to correct problems, you have held broken packages.
root@oc-pve-01:~# `

CrimsonKnight13, 2024-12-25:
qemu-server 8.3.2 is required for proxmox-ve & the rest.

Additional info:

The following packages have unmet dependencies:
 libpve-storage-perl : Breaks: qemu-server (< 8.3.2) but 8.3.0+port3 is to be installed
                       Recommends: pve-esxi-import-tools (>= 0.6.0) but it is not installable
E: Unable to correct problems, you have held broken packages.

apt search qemu-server
libpve-guest-common-perl 5.1.6 [/bookworm port]
└── Proxmox VE common guest-related modules

qemu-server 8.3.0+port3 [/bookworm port]
└── Qemu Server Tools

qemu-server-dbgsym 8.3.0+port3 [/bookworm port]
└── debug symbols for qemu-server

Alexr03, 2024-12-26:
Same here, same error with qemu-server cannot be installed.
Of course the day I wanted to try this, is the same day there's problems 😅

Hopefully can be resolved soon 🤞🏼

totheogreat, 2024-12-27:
Perfect tx! retried Installation, worked fine!

totheogreat, 2024-12-27:
completed!

closed 2024-12-27 by totheogreat