jiangcuo/pxvirt #152: Virtual machine does not see USB devices

pro100805, 2024-12-31:
hello, by running this command: sudo apt search qemu-server

I get the following response

Sorting... Done
Full Text Search... Done
libpve-guest-common-perl/unknown,now 5.1.6 all [installed,automatic]
Proxmox VE common guest-related modules

qemu-server/unknown,now 8.3.3+port2 arm64 [installed,automatic]
Qemu Server Tools

qemu-server-dbgsym/unknown 8.3.3+port2 arm64
debug symbols for qemu-server

and : qemu-system-arm --version

I get the following response

QEMU emulator version 9.2.0 (pve-qemu-kvm_9.2.0-1+port2)
Copyright (c) 2003-2024 Fabrice Bellard and the QEMU Project developers

virtual machine with router (and WIN 11) do not see USB devices, with virtual machine on Ubuntu everything is fine
tell me what I need to do

pro100805, 2024-12-31:
Hi, i have a virtual computer with an arm 64 router. In version 8.2.7 the modem connected via USB worked fine. After the update the router stopped seeing the USB modem. The hardware remains unchanged

jiangcuo, 2025-01-02:
https://github.com/jiangcuo/qemu-server/blob/e48ebea4570c64547cc1747523f6032e4d68e621/PVE/QemuServer/PCI.pm#L181

You can modify the bus number of this xhci to 0 and use an available addr.
```
'xhci' => { bus => 0,addr => 17 }         /usr/share/perl5/PVE/QemuServer/PCI.pm

```

pro100805, 2025-01-02:
having made changes in the PCI.pm file, the following message appears when trying to start the virtual machine

kvm: -device qemu-xhci,p2=15,p3=15,id=xhci,bus=pcie.0,addr=0x1: PCI: slot 1 function 0 not available for qemu-xhci, in use by usb-ehci,id=ehci
TASK ERROR: start failed: QEMU exited with code 1

jiangcuo, 2025-01-02:
Use addr=17 or other free

pro100805, 2025-01-02:
thanks, I have selected the following values and everything works 

'xhci' => { bus => 0, addr => 30 }

closed 2025-04-23 by jiangcuo