jiangcuo/pxvirt #153: The compiled PVE image cannot virtualize Windows 10/11 ARM on Kunpeng 920

Sakuraofking, 2025-01-02:
**Describe the bug**
I compiled the installation image of PVE 8.3.2+port using [pve-iso-builder](https://github.com/jiangcuo/pve-iso-builder) and installed it on the Kunpeng 920 server. Other ARM 64 architecture Linux distributions work well. But black screen occurs when installing Windows 10 ARM virtual machine.

**To Reproduce**
1. Create VM.
[image]
2. GPU choose virtio-gpu. Other defaults，BIOS choose OVMF.
[image]
3. SCSI disk adjust to a larger capacity.
[image]
4. CPU Cores 16. Type default.
[image]
5. Memory 16G.
[image]
6. Choose start, it can be start.
[image]
7. It can see press any key. But it doesn't seem to work when typing on the keyboard.
[image]
8. Balck screen
[image]

**Screenshots**
Hardware is this
[image]
Node info. Notice: raspberry is this server hostname,not raspberrypi :)
[image]

**ENV (please complete the following information):**
package version
[image]
[image]
CPU & OS
[image]

**Additional context**
Add any other context about the problem here.

jiangcuo, 2025-01-02:
Try wiki first

Sakuraofking, 2025-01-02:
Do you mean I need to install the debian12 operating system on the server? After that, should I add software sources to the operating system and install PVE? It has been tested and is the same as the above situation.
[image]
[image]
Is it a problem with installing the Windows ISO? But I have already replaced many Windows 10/11 ARM installation images.

Sakuraofking, 2025-01-03:
@jiangcuo Is there any other way to try?

chubirdh, 2025-01-06:
VGA should be ramfb first。 vga can be change to virt-GPU when Windows is installed and virtgpudo driver is installed.

Sakuraofking, 2025-01-06:
[image]
[image]
@chubirdh Hi, there.

chubirdh, 2025-01-07:
Windows 11 22H2 can be installed, but Windows 10 and Windows 11 24H2 can't be installed.
[image]
[image]

Sakuraofking, 2025-01-08:
Thanks. This question is solved.

closed 2025-01-08 by Sakuraofking