jiangcuo/pxvirt #166: 部署 PVE 7.4.1 和 8.3.1 时初始化 ramdisk 遇到 EFI_RNG_PROTOCOL unavailable 报错

ljysdfz, 2025-02-11:
**Describe the bug**
A clear and concise description of what the bug is.
我的联想飞天M740J使用的芯片架构是 ARM64 的 D2000，UEFI 使用中电科技的昆仑固件，现在在部署 Proxmox 的时候启动安装程序过程出现了 EFI Stub：EFI_RNG_PROTOCOL unavailable 问题。具体的回显如下：

EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable
EFI stub: Using DTB from configuration table
EFI stub:Exiting boot seruices...

**To Reproduce**
Steps to reproduce the behavior:
1. 制作启动 UBOOT
2. UEFI 设置 U 盘优先启动
3. 插入 U 盘启动进入安装界面，进入高级设置里的 debug 安装
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
原本应当在加载 inird.img 后正常进入安装程序

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**ENV (please complete the following information):**
 - OS: 麒麟 v10 内核 5.4.18 使用昆仑固件 UEFI
 - ARCH: arm64，芯片为飞腾 D2000
 - Version: 尝试了 7.4.1 和 8.3.1 都遇到这个问题

**Additional context**
Add any other context about the problem here.

1. 我在自带的麒麟系统里做了一些尝试，似乎并没有硬件 RNG，因此 rng-tools 不可用，但是软 rng 也就是 haveged 是系统预置了的。
很可能是昆仑固件本身禁掉了 RNG 接口

2. 试着修改 proxmox 安装程序的 GRUB 启动项，在 linux 条目添加 RANDOM.TRUST_CPU RANDOM.TRUST_BOOTLOADER 没有效果

3. 不知道是不是需要重新编译 iso 镜像，为其加入 arm64 架构的 haveged 作为软随机数生成器

jiangcuo, 2025-02-12:
请提供内核打印日志，或者从debian安装proxmox

ljysdfz, 2025-02-14:
> 请提供内核打印日志，或者从debian安装proxmox

您好 我当时也下载了 [debian 的 ARM 架构系统](https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/) 进行对照试验，发现安装时报错提示一样，证明主要问题在于系统无法正常安装，目测是昆仑固件的限制。

看报错的内容，似乎是有个标准的随机数接口调不到，不知道是不是昆仑 UEFI 直接默认使用 secure boot，目前也无法从昆仑方面得到固件更新可以提供这个接口。

猜测也可以在 iso 加上软随机数生成器，例如 haveged，直接在 initrd 启动内核的时候启动内部的软随机数生成器。

以上仅是我个人的推断，我先去想办法打印内核日志，再提供上来。

Thanks in advance。

ljysdfz, 2025-02-14:
> 请提供内核打印日志，或者从debian安装proxmox

您好 我在 GRUB 尝试了很多不同的手段，包括 

- debian 安装界面选择 recovery 模式
- linux 启动选项进入 recovery 模式
- linux 启动选项输出详细内核调试信息

但是均无法进入安装文件的内核查看日志

这里我贴出进入我的信创电脑的原生麒麟系统的 [dmesg](https://github.com/user-attachments/files/18797712/dmesg.txt)仅供参考

如有其他调试手段，请不吝赐教。

Thanks in advance。

ljysdfz, 2025-02-14:
> 不知道是不是昆仑 UEFI 直接默认使用 secure boot

BTW，这个地方我说错了，我刚注意到 UEFI 中是有 secure boot 选项的，并且我已经关闭了。

jiangcuo, 2025-02-14:
请直接联系服务器厂家吧，某些无法装麒麟以外的系统

ljysdfz, 2025-02-14:
> 请直接联系服务器厂家吧，某些无法装麒麟以外的系统

好的

closed 2025-02-18 by ljysdfz