jiangcuo/pxvirt #167: Backups to Proxmox Backup Server do not work

FredK27, 2025-02-12:
**Describe the bug**
I have the latest version of the Proxmox for the Raspberry Pi (arm64) build installed on three Raspberry Pi 5 16 GB boards to create an HA cluster. I have Proxmox Backup Server running externally on another x86 Proxmox cluster. I have set up access to PBS inside the Raspberry Pi cluster. I can run one backup of running VMs and complete the Backup successfully. All attempts to create additional backups after the initial one failed with the following error -

ERROR: VM 401 qmp command 'backup' failed - backup register image failed: command error: no previous backup found, cannot do incremental backup

Here's the full run log -

INFO: starting new backup job: vzdump 401 --notes-template '{{guestname}} - Test incremental BU.' --node rpi-pve3 --remove 0 --storage RPI-backups --notification-mode auto --mode suspend
INFO: Starting Backup of VM 401 (qemu)
INFO: Backup started at 2025-02-12 13:49:26
INFO: status = running
INFO: backup mode: suspend
INFO: ionice priority: 7
INFO: VM Name: Debian-Server
INFO: include disk 'scsi0' 'RPI-storage:401/vm-401-disk-1.qcow2' 32G
INFO: include disk 'efidisk0' 'RPI-storage:401/vm-401-disk-0.qcow2' 64M
INFO: suspending guest
INFO: creating Proxmox Backup Server archive 'vm/401/2025-02-12T18:49:26Z'
INFO: skipping guest-agent 'fs-freeze', agent configured but not running?
ERROR: VM 401 qmp command 'backup' failed - backup register image failed: command error: no previous backup found, cannot do incremental backup
INFO: aborting backup job
INFO: resuming VM again after 4 seconds
INFO: resume vm
ERROR: Backup of VM 401 failed - VM 401 qmp command 'backup' failed - backup register image failed: command error: no previous backup found, cannot do incremental backup
INFO: Failed at 2025-02-12 13:49:31
INFO: Backup job finished with errors
TASK ERROR: job errors

Also, restores from the initial good backup do not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Run an initial backup to a properly configured PBS backup store. Suspend the VM being backed up during execution.
2. Note that the initial backup will appear to complete correctly.
3. Run a second backup of the same VM to the same PBS backup store. You will get the error above.
4. Atempts to restore the VM from the backup in step 1 will fail with similar errors.

**Expected behavior**
The second and subsequent backups should all work correctly. It should also be possible to restore the target VM from any of the successful backups.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: Debian 1:6.6.74-1+rpt1 (2025-01-27) 
 - ARCH: aarch64
 - Version [ do `pveversion -v`]

proxmox-ve: 8.3.0 (running kernel: 6.6.74+rpt-rpi-2712)
pve-manager: 8.3.3 (running version: 8.3.3/bbba1c53a1a65b24)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2+deb12u1
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx11
libjs-extjs: 7.0.0-5
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.1
libproxmox-rs-perl: 0.3.4
libpve-access-control: 8.2.0
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.10
libpve-cluster-perl: 8.0.10
libpve-common-perl: 8.2.9+port1
libpve-guest-common-perl: 5.1.6
libpve-http-server-perl: 5.1.2
libpve-network-perl: 0.10.0+port
libpve-rs-perl: 0.9.0
libpve-storage-perl: 8.3.3
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-4
proxmox-backup-client: 3.3.2-2
proxmox-backup-file-restore: 3.3.2-2
proxmox-firewall: 0.5.0
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.6
proxmox-widget-toolkit: 4.3.4
pve-cluster: 8.0.10
pve-container: 5.2.2+port1
pve-docs: 8.3.1
pve-edk2-firmware: not correctly installed
pve-firewall: 5.1.0
pve-firmware: not correctly installed
pve-ha-manager: 4.0.6
pve-i18n: 3.2.4
pve-qemu-kvm: 9.2.0-2
pve-xtermjs: 5.3.0-3
qemu-server: 8.3.7
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.9.0+port
vncterm: 1.8.0
zfsutils-linux: 2.2.7-pve1

**Additional context**
Backups to non-PBS (ex NFS or local stores) work correctly.

jiangcuo, 2025-02-13:
Backup vm  to arm64 backupserver works fine.
[image]
Make sure it is consistent with the following questions
https://github.com/jiangcuo/Proxmox-Port/issues/6

FredK27, 2025-02-13:
You don't understand the configuration here. Proxmox Backup Server is running as a VM on an x85 Proxmox cluster. I checked the PAGESIZE for the Server in the x86 environment, which is 4K.

The Source of the backups is a different Proxmox cluster built from three 16 GB Raspberry Pi 5 systems. I did the suggested reinstall of the following on the Raspberry Pi machines -

# apt reinstall proxmox-backup-file-restore proxmox-backup-restore-image

I just tried the backup and restore operations from my Raspberry Pi cluster and got the same errors. Note the the disks for the VMs on the Raspberry Pi that are being backup up use and NFS volume for storage.

FredK27, 2025-02-13:
Here is the configuration of one of the VMs on the Raspberry Pi Cluster that is failing backups to PBS -

[image]

jiangcuo, 2025-07-20:
There is not enough information to reproduce the issue, so the case is closed

closed 2025-07-20 by jiangcuo

FredK27, 2025-07-20:
This is very simple to reproduce. I can do it here 100% of the time. Here are the steps:

1) Set up a 3-node Raspberry Pi (RPi) based proxmox cluster
2) Install Proxmox Backup Server (PBS) as a VM on a different AMD64 (Intel) based cluster. Configure PBS to use an NFS share as the target backup datastore
3) Create a Debian VM on the RPi cluster.
4) Run an initial backup using Snapshot mode (it will work)
5) Run a second backup using Snapshot mode (it will fail as follows) -

()
INFO: starting new backup job: vzdump 400 --notes-template '{{guestname}}' --notification-mode auto --node rpi-pve1 --storage RPI-backups --mode snapshot --remove 0
INFO: Starting Backup of VM 400 (qemu)
INFO: Backup started at 2025-07-20 07:56:29
INFO: status = running
INFO: VM Name: Debian-Desktop
INFO: include disk 'scsi0' 'RPI-storage:400/vm-400-disk-1.qcow2' 64G
INFO: include disk 'efidisk0' 'RPI-storage:400/vm-400-disk-0.qcow2' 64M
INFO: backup mode: snapshot
INFO: ionice priority: 7
INFO: creating Proxmox Backup Server archive 'vm/400/2025-07-20T11:56:29Z'
INFO: issuing guest-agent 'fs-freeze' command
INFO: issuing guest-agent 'fs-thaw' command
ERROR: VM 400 qmp command 'backup' failed - backup register image failed: command error: no previous backup found, cannot do incremental backup
INFO: aborting backup job
INFO: resuming VM again
ERROR: Backup of VM 400 failed - VM 400 qmp command 'backup' failed - backup register image failed: command error: no previous backup found, cannot do incremental backup
INFO: Failed at 2025-07-20 07:56:29
INFO: Backup job finished with errors
TASK ERROR: job errors

Here is the configuration of the RPi-based Debian VM that is being backed up -

[image]

I have other VMs running the Deian server, and they exhibit the same failures.

Note that this same configuration of PBS supports two on AMD64 proxmox clusters with multiple similar backups daily.

If you take the time to set up the proper configuration of PBS on an AMD64 Proxmox Cluster and a Proxmox cluster on a Raspberry Pi, you will not encounter any issues reproducing this problem.

FredK27, 2025-07-20:
Please reopen and fix this PR.