jiangcuo/pxvirt #168: Upgrading libpve-common-perl to 8.2.9+port1 breaks web interface on Raspberry Pi

dezzak, 2025-02-13:
**Describe the bug**

Doing an apt upgrade recently, the web UI failed to come up.
Downgrading libpve-common-perl from 8.2.9+port1 to 8.2.9+port fixed the issue.

**To Reproduce**
Steps to reproduce the behavior:

1. apt upgrade
2. reboot
3. no UI

Checking the logs revealed perl compilation issues

```
Feb 13 09:35:38 pia systemd[1]: Starting pveproxy.service - PVE API Proxy Server...
░░ Subject: A start job for unit pveproxy.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit pveproxy.service has begun execution.
░░ 
░░ The job identifier is 2249.
Feb 13 09:35:39 pia pveproxy[6062]: Attempt to reload PVE/QemuServer.pm aborted.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/share/perl5/PVE/HA/Resources/PVEVM.pm line 14.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/HA/Resources/PVEVM.pm line 21.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/share/perl5/PVE/HA/Env/PVE2.pm line 23.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/HA/Env/PVE2.pm line 23.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/share/perl5/PVE/API2/Cluster.pm line 15.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/API2/Cluster.pm line 15.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/share/perl5/PVE/API2.pm line 15.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/API2.pm line 15.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/share/perl5/PVE/Service/pveproxy.pm line 13.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Service/pveproxy.pm line 13.
Feb 13 09:35:39 pia pveproxy[6062]: Compilation failed in require at /usr/bin/pveproxy line 11.
Feb 13 09:35:39 pia pveproxy[6062]: BEGIN failed--compilation aborted at /usr/bin/pveproxy line 11.
Feb 13 09:35:39 pia systemd[1]: pveproxy.service: Control process exited, code=exited, status=255/EXCEPTION
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit pveproxy.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 255.
Feb 13 09:35:39 pia systemd[1]: pveproxy.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit pveproxy.service has entered the 'failed' state with result 'exit-code'.
Feb 13 09:35:39 pia systemd[1]: Failed to start pveproxy.service - PVE API Proxy Server.
░░ Subject: A start job for unit pveproxy.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit pveproxy.service has finished with a failure.
░░ 
░░ The job identifier is 2249 and the job result is failed.

```

Checking what apt had changed (/var/log/apt/history.log):

```
Start-Date: 2025-02-13  08:42:27
Requested-By: dezza (1000)
Install: python3-ruamel.yaml:arm64 (0.17.21-1, automatic), libonig5:armhf (6.9.8-1, automatic), python3-xmltodict:arm64 (0.13.0-1, automatic), python3-ruamel.yaml.clib:arm64 (0.2.7-1+b2, automatic), jc:arm64 (1.22.5-1, automatic), jq:armhf (1.6-2.1, automatic), python3-pygments:arm64 (2.14.0+dfsg-1, automatic), libjq1:armhf (1.6-2.1, automatic)
Upgrade: proxmox-backup-restore-image:arm64 (0.6.1+port1, 0.6.1+port3), proxmox-widget-toolkit:arm64 (4.3.2, 4.3.4), libtasn1-6:arm64 (4.19.0-2, 4.19.0-2+deb12u1), pve-qemu-kvm:arm64 (9.2.0-1+port3, 9.2.0-2), raspi-config:arm64 (20250203, 20250210), libcamera-ipa:arm64 (0.4.0+rpt20250127-3, 0.4.0+rpt20250211-1), libcamera0.4:arm64 (0.4.0+rpt20250127-3, 0.4.0+rpt20250211-1), proxmox-backup-file-restore:arm64 (3.3.0-2, 3.3.2-2), qemu-server:arm64 (8.3.3+port2, 8.3.7), proxmox-backup-client:arm64 (3.3.0-2, 3.3.2-2), pve-manager:arm64 (8.3.2+port, 8.3.3), libpve-common-perl:arm64 (8.2.9+port, 8.2.9+port1)
Remove: linux-headers-6.6.51+rpt-rpi-v8:arm64 (1:6.6.51-1+rpt3), linux-kbuild-6.6.51+rpt:arm64 (1:6.6.51-1+rpt3), linux-headers-6.6.51+rpt-rpi-2712:arm64 (1:6.6.51-1+rpt3), linux-headers-6.6.51+rpt-common-rpi:arm64 (1:6.6.51-1+rpt3)
End-Date: 2025-02-13  08:43:37
```

Suspecting a perl issue, I noted `libpve-common-perl:arm64 (8.2.9+port, 8.2.9+port1)`

Rolling back this package `apt install libpve-common-perl=8.2.9+port` then restarting pveproxy fixed the issue

**Expected behavior**

Web UI comes up as normal

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: debian 12
 - ARCH: aarch64 / arm64 proxmox-port
 - Version 

```
proxmox-ve: 8.3.0 (running kernel: 6.6.74+rpt-rpi-2712)
pve-manager: 8.3.3 (running version: 8.3.3/bbba1c53a1a65b24)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.7-pve3
corosync: 3.1.7-pve3
criu: 3.17.1-2+deb12u1
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx11
libjs-extjs: 7.0.0-5
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.1
libproxmox-rs-perl: 0.3.4
libpve-access-control: 8.2.0
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.10
libpve-cluster-perl: 8.0.10
libpve-common-perl: 8.2.9+port1
libpve-guest-common-perl: 5.1.6
libpve-http-server-perl: 5.1.2
libpve-network-perl: 0.10.0+port
libpve-rs-perl: 0.9.0
libpve-storage-perl: 8.3.3
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-4
proxmox-backup-client: 3.3.2-2
proxmox-backup-file-restore: 3.3.2-2
proxmox-firewall: 0.5.0
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.6
proxmox-widget-toolkit: 4.3.4
pve-cluster: 8.0.10
pve-container: 5.2.2+port1
pve-docs: 8.3.1
pve-edk2-firmware: not correctly installed
pve-firewall: 5.1.0
pve-firmware: not correctly installed
pve-ha-manager: 4.0.6
pve-i18n: 3.2.4
pve-qemu-kvm: 9.2.0-2
pve-xtermjs: 5.3.0-3
qemu-server: 8.3.7
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.9.0+port
vncterm: 1.8.0
zfsutils-linux: 2.2.7-pve1
```

**Additional context**
None yet

jiangcuo, 2025-02-15:
Please provide more logs, this log can not determine the problem

dezzak, 2025-02-17:
Running pveprox with 8.2.9+port:
```
root@pia:~# apt-cache policy libpve-common-perl
libpve-common-perl:
  Installed: 8.2.9+port
  Candidate: 8.2.9+port1
  Version table:
     8.2.9+port1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
 *** 8.2.9+port 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
        100 /var/lib/dpkg/status
     8.2.9 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.2.6 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.2.3 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.2.1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.1.1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.1.0-1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.10 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.9 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.8 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.6 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.5 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
root@pia:~# pveproxy status
stopped
```
pveproxy runs and returns a status

Now install 8.2.9+port1 and try the same
```
root@pia:~# apt install libpve-common-perl=8.2.9+port1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be upgraded:
  libpve-common-perl
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/118 kB of archives.
After this operation, 1,024 B of additional disk space will be used.
Reading changelogs... Done
(Reading database ... 99714 files and directories currently installed.)
Preparing to unpack .../libpve-common-perl_8.2.9+port1_arm64.deb ...
Unpacking libpve-common-perl (8.2.9+port1) over (8.2.9+port) ...
Setting up libpve-common-perl (8.2.9+port1) ...
Processing triggers for pve-manager (8.3.3) ...
Job for pvedaemon.service failed.
See "systemctl status pvedaemon.service" and "journalctl -xeu pvedaemon.service" for details.
Job for pvestatd.service failed.
See "systemctl status pvestatd.service" and "journalctl -xeu pvestatd.service" for details.
Job for pvescheduler.service failed.
See "systemctl status pvescheduler.service" and "journalctl -xeu pvescheduler.service" for details.
Processing triggers for pve-ha-manager (4.0.6) ...
root@pia:~# pveproxy status
Attempt to reload PVE/QemuServer.pm aborted.
Compilation failed in require at /usr/share/perl5/PVE/HA/Resources/PVEVM.pm line 14.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/HA/Resources/PVEVM.pm line 21.
Compilation failed in require at /usr/share/perl5/PVE/HA/Env/PVE2.pm line 23.
[... 131 lines cut ...]
}

print(Dumper(read_cpuinfo()));
```

and get the results:

```
Before if statement:
$VAR1 = 'aarch64';
$VAR2 = 'unknown';
$VAR3 = 1;
Contents of $hardinfo:
$VAR1 = '';
malformed JSON string, neither tag, array, object, number, string or atom, at character offset 0 (before "(end of string)") at test.pl line 56, <GEN1> line 1.
```

So it appears that the `decode_json()` call doesn't like being given an empty string

Indeed, it does appear that `/run/pve/query-machine-info` is empty:

```
# ls -l /run/pve/query-machine-info
-rw-r--r-- 1 root root 0 Feb 13 14:38 /run/pve/query-machine-info
```

jiangcuo, 2025-02-17:
In the new version, I use dmidecode to get hardware information. Maybe there is an error in this part on sbc. I will fix this problem in the morning of 8th East

jiangcuo, 2025-02-18:
Can you run this script? /usr/bin/query-machine-info
I haven't been able to reproduce this issue on the Rockchip SBC. Have you installed dmidecode and jc?

jiangcuo, 2025-02-18:
Would yout be convenient to provide the complete output of `cat /proc/cpuinfo` and `lscpu`?

dezzak, 2025-02-18:
>Can you run this script? /usr/bin/query-machine-info

It errors while running:

```
# /usr/bin/query-machine-info 
jq: error while loading shared libraries: libm.so.6: ELF load command address/offset not page-aligned
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
BrokenPipeError: [Errno 32] Broken pipe
```

> Have you installed dmidecode and jc?

I've not explicitly installed them, but they are installed:

```
$ apt-cache policy dmidecode
dmidecode:
  Installed: 3.4-1
  Candidate: 3.4-1
  Version table:
 *** 3.4-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
$ apt-cache policy jc
jc:
  Installed: 1.22.5-1
  Candidate: 1.22.5-1
  Version table:
 *** 1.22.5-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
        500 http://deb.debian.org/debian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
```

> Would yout be convenient to provide the complete output of cat /proc/cpuinfo and lscpu?

```
$ cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

Revision	: d04170
Serial		: a1d26b9c00626123
Model		: Raspberry Pi 5 Model B Rev 1.0
```

```
$ lscpu
Architecture:             aarch64
  CPU op-mode(s):         32-bit, 64-bit
  Byte Order:             Little Endian
CPU(s):                   4
  On-line CPU(s) list:    0-3
Vendor ID:                ARM
  Model name:             Cortex-A76
    Model:                1
    Thread(s) per core:   1
    Core(s) per cluster:  4
    Socket(s):            -
    Cluster(s):           1
    Stepping:             r4p1
    CPU(s) scaling MHz:   62%
    CPU max MHz:          2400.0000
    CPU min MHz:          1500.0000
    BogoMIPS:             108.00
    Flags:                fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
Caches (sum of all):      
  L1d:                    256 KiB (4 instances)
  L1i:                    256 KiB (4 instances)
  L2:                     2 MiB (4 instances)
  L3:                     2 MiB (1 instance)
NUMA:                     
  NUMA node(s):           1
  NUMA node0 CPU(s):      0-3
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Not affected
  Retbleed:               Not affected
  Spec rstack overflow:   Not affected
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:             Mitigation; __user pointer sanitization
  Spectre v2:             Mitigation; CSV2, BHB
  Srbds:                  Not affected
  Tsx async abort:        Not affected
```

dezzak, 2025-02-18:
I wonder if this is due having to specify the arch in /etc/apt/sources.list.d/pveport.list?

```
$ apt-cache policy jq
jq:
  Installed: (none)
  Candidate: 1.6-2.1
  Version table:
     1.6-2.1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
$ which jq
/usr/bin/jq
$ dpkg -S /usr/bin/jq
jq:armhf: /usr/bin/jq
$ apt-cache policy jq:armhf
jq:armhf:
  Installed: 1.6-2.1
  Candidate: 1.6-2.1
  Version table:
 *** 1.6-2.1 500
        500 http://deb.debian.org/debian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
```

jiangcuo, 2025-02-18:
It seems that armhf's jq uses arm64 libraries, causing incompatibility issues, not my code.
But this also needs me to fix, to avoid such a situation. You can install jq:arm64
I don't understand why the Raspberry PI still uses armhf, even though it is hardware supported

dezzak, 2025-02-24:
Sorry for the delay - I've been away.

`apt install jq:arm64` was successful, as was:

```
# /usr/bin/query-machine-info
# cat /run/pve/query-machine-info
[
  {
    "cpu": "Cortex-A76"
  }
]
```
Installing the latest pveproxy and restarting the service is also now successful:
```
# apt install libpve-common-perl=8.2.9+port1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libjq1:armhf libonig5:armhf
Use 'apt autoremove' to remove them.
The following packages will be upgraded:
  libpve-common-perl
1 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Need to get 0 B/118 kB of archives.
After this operation, 1,024 B of additional disk space will be used.
Reading changelogs... Done
(Reading database ... 99718 files and directories currently installed.)
Preparing to unpack .../libpve-common-perl_8.2.9+port1_arm64.deb ...
Unpacking libpve-common-perl (8.2.9+port1) over (8.2.9+port) ...
Setting up libpve-common-perl (8.2.9+port1) ...
Processing triggers for pve-manager (8.3.3) ...
Processing triggers for pve-ha-manager (4.0.6) ...
# systemctl restart pveproxy
# pveproxy status
running
```

The perl script could be adjusted to check for an empty file and treat it the same as if the file didn't exist, and it probably should. But I guess the wider issue as around the architecture being messed up somewhere.

I believe I followed [this guide](https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm) that suggests doing
`echo 'deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port'>/etc/apt/sources.list.d/pveport.list`, but I also recall that somewhere in the process this was modified by dpkg which then broke further updates

According to my apt logs, I ran the following to install proxmox:

```
Start-Date: 2025-01-23  10:47:22
Commandline: apt install proxmox-ve postfix open-iscsi
```

```
Setting up pve-manager (8.3.2+port) ...

Configuration file '/etc/apt/sources.list.d/pveport.list'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** pveport.list (Y/I/N/O/D/Z) [default=N] ? D
ESC[?1hESC=^M--- /etc/apt/sources.list.d/pveport.list        2025-01-23 10:36:16.733205714 +0000ESC[m
+++ /etc/apt/sources.list.d/pveport.list.dpkg-new       2024-12-26 21:23:21.000000000 +0000ESC[m
@@ -1 +1 @@ESC[m
-deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm portESC[m
+deb https://mirrors.apqa.cn/proxmox/debian/pve bookworm portESC[m
\ No newline at end of fileESC[m
^MESC[KESC[?1lESC>
Configuration file '/etc/apt/sources.list.d/pveport.list'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** pveport.list (Y/I/N/O/D/Z) [default=N] ? I
Installing new version of config file /etc/apt/sources.list.d/pveport.list ...
update successful
```

I recall that the next time I tried to do an apt update, things went wrong, so I reverted back

closed 2025-04-23 by jiangcuo