jiangcuo/pxvirt #171: where is the listed iso for the rockchip install

jpdacostaza, 2025-03-02:
[image]

jpdacostaza, 2025-03-02:
[image]

double and trippel checked i can not find it.. please advise

elaldradrian, 2025-03-09:
Just not maintained anymore - see: https://github.com/jiangcuo/Proxmox-Port/issues/137

I've installed it with a vanilla Debian bookworm and added the repo afterwards. :)

jpdacostaza, 2025-03-09:
Hey thanks for sharing EFI bootinis working fine with spi flash.
What kernel is your proxmox on after the debian route.
Want to know if there is a difference between the kernels so I'm testing now

jpdacostaza, 2025-03-09:
> Just not maintained anymore - see: [#137](https://github.com/jiangcuo/Proxmox-Port/issues/137)
> 
> I've installed it with a vanilla Debian bookworm and added the repo afterwards. :)

Also sorry what are you running it on..?

jpdacostaza, 2025-03-09:
https://github.com/jiangcuo/Proxmox-Port/issues/137

closed 2025-03-09 by jpdacostaza