jiangcuo/pxvirt #177: Proxmox VE 8.4

x13-me, 2025-04-23:
Proxmox VE 8.4 was released 2025-04-09, no build is currently available, nor have updated packages been uploaded to the apt repo.

This release adds several key Quality-of-Life improvements, along with upgrading many dependencies.

See [Roadmap#8.4-known-issues](https://pve.proxmox.com/wiki/Roadmap#8.4-known-issues)

- Based on Debian Bookworm (12.10)
- Latest 6.8.12-9 Kernel as new stable default
- Newer 6.14 Kernel as opt-in
- QEMU 9.2.0
- LXC: 6.0.0
- ZFS: 2.2.7 (with compatibility patches for Kernel 6.14)
- Ceph Squid 19.2.1
- Ceph Reef 18.2.4
- Ceph Quincy 17.2.8 is end-of-life, users are advised to upgrade

x13-me, 2025-04-23:
@jiangcuo do we have an ETA on this?

LeenHawk, 2025-05-15:
also ask

LeenHawk, 2025-05-16:
> It seems that the project got renamed, the logo aswell , and to get to ver 8.4.x you should change the repo as the documentation is saying. It's in Chinese, but with google translate you should be ok.

大哥，在哪啊。。。文档没看到啊

milumil, 2025-05-16:
> > It seems that the project got renamed, the logo aswell , and to get to ver 8.4.x you should change the repo as the documentation is saying. It's in Chinese, but with google translate you should be ok.
> 
> 大哥，在哪啊。。。文档没看到啊

Please keep it in english.
https://github.com/jiangcuo/pxvirt

[image]

x13-me, 2025-07-06:
Closing as no longer relevant

closed 2025-07-06 by x13-me