jiangcuo/pxvirt #180: Memory usage of a LXC container shows 0

austonpramodh, 2025-05-02:
Hi,

I am hitting a wierd bug, not sure where to debug, 

This could be a CT issue:

https://www.reddit.com/r/Proxmox/comments/1cq2twj/lxc_alpine_memory_usage_shows_0gb/

I did try to pull different containers, the latest one I tried was from https://github.com/jiangcuo/Proxmox-Port/issues/174#issuecomment-2790777382

[image]

I will try with VM too and report.

austonpramodh, 2025-05-02:
[image]

Looks like theres no issue with VM, its on Containers

spiritLHLS, 2025-05-09:
Have you try any of https://github.com/oneclickvirt/lxc_arm_images/releases ?

austonpramodh, 2025-05-10:
[image]

Tried [alpine_3.21_3.21_arm64_cloud.tar.xz](https://github.com/oneclickvirt/lxc_arm_images/releases/download/alpine/alpine_3.21_3.21_arm64_cloud.tar.xz), same issue.

[image]

austonpramodh, 2025-05-10:
Tried with fresh install, Still no usage shown

austonpramodh, 2025-05-11:
[image]

Finally found the issue.

Raspian lite OS doesn't enable cgroups memory controller, had to add enable it by add `cgroup_enable=memory` to `/boot/firmware/cmdline.txt`.

closed 2025-05-11 by austonpramodh