jiangcuo/pxvirt #184: The gpg key used to sign packages has seemingly changed, and I can't figure out where to find it.

AdrianKoshka, 2025-06-24:
Your guide says to import the gpg key, which I tried doing with no success. Instead, apt complains about a public key with a completely different ID.

```
root@xc-ls:/etc/apt# apt update
Hit:1 https://download.lierfang.com/pxcloud/pxvirt bookworm InRelease                                         
Hit:2 https://mirrors.lierfang.com/debian-ports/debian sid InRelease
Err:2 https://mirrors.lierfang.com/debian-ports/debian sid InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7F00470E9E4625DC
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://mirrors.lierfang.com/debian-ports/debian sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7F00470E9E4625DC
W: Failed to fetch https://mirrors.lierfang.com/debian-ports/debian/dists/sid/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7F00470E9E4625DC
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
I can't find this public key, and therefore can't update my system.

jiangcuo, 2025-06-25:
What platform are you using? This debian-ports is only applicable to Loongarch. If you are using Loongarch, you can ignore this issue.

AdrianKoshka, 2025-06-25:
I'm using Loongarch yes, and no I can't ignore it, since it breaks updates.

jiangcuo, 2025-06-27:
```
deb [ allow-insecure=yes ] http://mirrors.lierfang.com/debian-ports/debian sid main
```

jiangcuo, 2025-07-20:
No update for a long time, closed