jiangcuo/pxvirt #190: repo gpg key is blank and mirrors do not have release file

7thgenerationdesign, 2025-07-29:
When I import the gpg key (`curl -L https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o /etc/apt/trusted.gpg.d/pveport.gpg`), the imported file contains nothing except contact info for pveport:

[image]

This obviously causes problems when I update:

[image]

Also, I'm getting the error that `The repository 'https://mirrors.apqa.cn/proxmox/debian/pve bookworm Release' does not have a Release file.`. I tried all other mirrors listed here, none work.

Thank you for all of the hard work developing this thus far!

tishimself, 2025-07-30:
+1. Been checking for an update the last 4 days or so. Tried all the previously listed mirrors.

sebaxtian, 2025-08-14:
Hello everyone,

The issue currently exists with the repositories listed in the existing documentation here:
https://github.com/jiangcuo/pxvirt/wiki/Proxmox%E2%80%90Port--Repo-List

[image]

---

I have two RPis, an RPi 4 and an RPi 5, both running Debian Bookworm.

```bash
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

---

Follow these steps to resolve the issue using the PXVIRT mirror:
https://docs.pxvirt.lierfang.com/en/installfromdebian.html

**Step 1**
Delete any conflicting repository configuration files you currently have. For example, in my case, I removed the following:

```bash
rm -fv /etc/apt/trusted.gpg.d/pveport.gpg
```

```bash
rm -fv /etc/apt/sources.list.d/pveport.list
```

**Step 2**
Add the new GPG key and the new repository to the sources list:

```bash
curl -L https://mirrors.lierfang.com/pxcloud/lierfang.gpg -o /etc/apt/trusted.gpg.d/lierfang.gpg
```

```bash
echo 'deb [arch=arm64] https://mirrors.lierfang.com/pxcloud/pxvirt bookworm main'>/etc/apt/sources.list.d/pxvirt-sources.list
```

**Step 3**
Finally, update and install the packages from the new mirror:

```bash
apt update
```

[image]

```bash
apt upgrade
```

[image]

---

That’s it. This is how I was able to solve the problem in my case, and I hope this solution can help more people. 🦾 

Best regards. 🚀

InWamos, 2025-08-17:
Changing 

> /etc/apt/sources.list.d/pxvirt-sourses.list

 Content to:
`deb [arch=arm64 trusted=yes] http://mirrors.lierfang.com/pxcloud/pxvirt bookworm main`
did the trick. **But be aware that this isn't safe to do this as the image downloaded from such repo won't be GPG verified**

agarmoli, 2025-08-17:
No working:

Ign:5 https://mirrors.lierfang.com/pxcloud/pxvirt bookworm InRelease
Err:6 https://mirrors.lierfang.com/pxcloud/pxvirt bookworm Release
  Redirection from https to 'http://de.mirrors.lierfang.com/pxcloud/pxvirt/dists/bookworm/Release' is forbidden [IP: 117.139.168.42 443]
Error: The repository 'https://mirrors.lierfang.com/pxcloud/pxvirt bookworm Release' does not have a Release file.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.

k5map, 2025-08-17:
Ditto... I followed the steps above using Bookworm and get the same error

k5map, 2025-08-20:
No change on my end... same error as previously reported which is a disappointment.  I'm prevented from building 4 new hosts for my cluster.

Red-Banana-Official, 2025-08-20:
I updated my 2 AMD64 proxmox servers to proxmox 9, which causes issues in the cluster with my raspberry pi. being at proxmox 8.
Reverting updates isn't really an option.

Hopefully they update it really soon.

gontadu, 2025-09-08:
Is this related to #205 too?