jiangcuo/pxvirt #201: Bookworm repository error

k5map, 2025-08-17:
While building a new PVE host, when I enter "apt update", I receive the following error:

Err:6 https://mirrors.lierfang.com/pxcloud/pxvirt bookworm Release
  Redirection from https to 'http://de.mirrors.lierfang.com/pxcloud/pxvirt/dists/bookworm/Release' is forbidden [IP: 117.139.168.42 443]
Reading package lists... Done
E: The repository 'https://mirrors.lierfang.com/pxcloud/pxvirt bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

If I proceed will have a good install of PXVIRT?

k5map, 2025-08-17:
This may be related to the above issue... Also, when I run "apt update" on one of my existing hosts, I get this error:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://download.lierfang.com/pxcloud/pxvirt bookworm InRelease' does not seem to provide it (sources.list entry misspelt?)

agarmoli, 2025-08-17:
Same here:

Hit:1 http://deb.debian.org/debian trixie InRelease
Hit:2 http://deb.debian.org/debian-security trixie-security InRelease
Hit:3 http://deb.debian.org/debian trixie-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian trixie InRelease
Ign:5 https://mirrors.lierfang.com/pxcloud/pxvirt trixie InRelease
Err:6 https://mirrors.lierfang.com/pxcloud/pxvirt trixie Release
  Redirection from https to 'http://de.mirrors.lierfang.com/pxcloud/pxvirt/dists/trixie/Release' is forbidden [IP: 117.139.168.42 443]
Error: The repository 'https://mirrors.lierfang.com/pxcloud/pxvirt trixie Release' does not have a Release file.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.

k5map, 2025-08-17:
I see you upgraded to Trixie.  Are you running PVE on a RPi?  If so, how did you upgrade to Trixie?

k5map, 2025-08-17:
This issue appears to be a duplicate [#190 ](https://github.com/jiangcuo/pxvirt/issues/190) so closing this issue